Ultimaker/Cura #12090: Cura 5.0 slice leads to stuttering tool head vs. 4.13 master does not - Klipper, Pi has lots of processing headroom.

PhilBaz, 2022-05-04:
### Application Version

5.0 vs 4.13 master branch

### Platform

win 10

### Printer

Voron - klipper - raspberry pi 3b

### Reproduction steps

Sliced the attached file in both versions with similar settings....5.0 slice tool head stutters (move, pause, move pause) even though Pi has plenty of processing headroom. Slice on 4.13 does not stutter. 

### Actual results

-

### Expected results

No stuttering from sliced file. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Here are the files in .zip to satisfy git. I've manually truncated the code half way through the files to remove layers.

[PLNTR_SPHR V7_Re_top test - Cura 5.zip](https://github.com/Ultimaker/Cura/files/8622270/PLNTR_SPHR.V7_Re_top.test.-.Cura.5.zip)
[PLNTR_SPHR V7_Re_top test - Master 4.13.gcode.zip](https://github.com/Ultimaker/Cura/files/8622272/PLNTR_SPHR.V7_Re_top.test.-.Master.4.13.gcode.zip)

smartavionics, 2022-05-04:
See https://github.com/Ultimaker/Cura/issues/6748

MariskaMaas, 2022-05-23:
Hi PhilBaz,

Thanks for reporting your issue. 
I see the gcode is displaying the stutter behavior you mentioned.
Can you share your curaproject instead? Than I can also see the settings you used to get to this point. 
There are some new behavior and settings introduced in Cura 5.0 that are related to thin walls, that might help you out. 

As mentioned in the last comment in https://github.com/Ultimaker/Cura/issues/6748 we have made some improvements between the 5.0 Beta and the 5.0 stable. Can you check again in the latest version of Cura 5.0?

PhilBaz, 2022-05-26:
@MariskaMaas 

I'm attaching the project file here...Hope this helps. It ended up printing perfectly with @smartavionics Cura branch.

[PLNTR_SPHR.V7_Re_top.test.-.Cura.5.zip](https://github.com/Ultimaker/Cura/files/8780632/PLNTR_SPHR.V7_Re_top.test.-.Cura.5.zip)

PhilBaz, 2022-06-28:
@MariskaMaas @smartavionics 

The most recent version of Cura still have the same stuttering issue...Ill attach files here.

Did you look at the solution provided by @smartavionics. His fork is working perfectly.

[stuttering.zip](https://github.com/Ultimaker/Cura/files/9004270/stuttering.zip)

nallath, 2022-07-05:
> Did you look at the solution provided by @smartavionics. His fork is working perfectly.

His fork doesn't yet have Arachne, which is essentially the main difference between 4.13 and 5.0. So The fact that it works there doesn't provide any extra info.

benjjyman, 2022-07-10:
I'm having stuttering on cylindrical items, its kind of edge case where using thick line widths in a small area. The nozzle speeds up and slows down travel (I assume to adjust flow rate, so as to eliminate the need for gap fill ?).

my stl's used before slicing.
[MyTensioningRodSpacer.zip](https://github.com/Ultimaker/Cur
[lowresMyTensioningRodSpacer.zip](https://github.com/Ultimaker/Cura/files/9079844/lowresMyTensioningRodSpacer.zip)
a/files/9079843/MyTensioningRodSpacer.zip)

cura profile (downloaded for a ender 2, but also repurposed for other machines)
[FFV5.0_ExtraFast(0.28).zip](https://github.com/Ultimaker/Cura/files/9079845/FFV5.0_ExtraFast.0.28.zip)

resulting gcode output.
[CE2_lowresMyTensioningRodSpacer.zip](https://github.com/Ultimaker/Cura/files/9079846/CE2_lowresMyTensioningRodSpacer.zip)

The stuttering can be seen in the preview if you watch the nozzle velocity, you will notice on some circles it runs smoothly then on thicker lines is stutters. 

Things tried:

Octoprint via usb
Sd card
Arc welder plugin

results:-

use thinner layer widths produces a smooth motion , leading to a smooth print.

PhilBaz, 2023-05-04:
Still happening in 5.3. pretty major bug.
[Phil_A_PLA_f.93_210C_200mm.gcode.zip](https://github.com/Ultimaker/Cura/files/11392680/Phil_A_PLA_f.93_210C_200mm.gcode.zip)

MariMakes, 2023-05-04:
Hey @PhilBaz,

Thanks for your report 👍
I took a look, it looks similar to this jagged paths on curved surfaces report; https://github.com/Ultimaker/Cura/issues/14811
It seems like Cura seems to be adding micro segments that are hard to process. Our developers are working on improving the behavior.

Do you mind adding the project file  to the linked ticket https://github.com/Ultimaker/Cura/issues/14811
It contains the printer and settings we need for troubleshooting. 
To save a project file go to File -> Save project.

PhilBaz, 2023-05-05:
@MariMakes Sure no problem.... Ill post the file there.

jellespijker, 2023-07-08:
Hi @PhilBaz can you confirm if the fix for https://github.com/Ultimaker/Cura/issues/14811#issuecomment-1621079258 
also solved your issues?

PhilBaz, 2023-07-12:
@jellespijker Yes, it did. I was participating in the thread there. Thank you!