Ultimaker/Cura #12091: Holes by design are filled

MatousRokosVolue, 2022-05-04:
### Application Version

5.0

### Platform

Windows 11

### Printer

Snapmaker A350

### Reproduction steps

Object with the hole from the bottom by design gets filled and sliced solid. 
Actual object: 
[image]

### Actual results

Sliced preview:
[image]

The hole is actually visible in preview yet filled anyway:
[image]

### Expected results

Holes should be printed as expected and rendered in sliced preview

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[S2A_bear-3.zip](https://github.com/Ultimaker/Cura/files/8626523/S2A_bear-3.zip)

smartavionics, 2022-05-04:
Interestingly, if you turn the model on its side, the hole appears!

[image]

GregValiant, 2022-05-04:
It's the infamous curse of "Make Overhangs Printable".  Turn that off as it is for extremely specific and not often encountered problems.
[image]

MatousRokosVolue, 2022-05-04:
yeah, that's it. Can confirm that "Make Overhangs Printable" setting does it in both 5.0 and 4.13.1. Still relevant as bug report perhaps :)

GregValiant, 2022-05-04:
It's "expected behavior".  The setting is one I've never had to use but I know it has helped with some problems (not many).
"Remove all Holes" is another one.  New users seem to figure it's a good idea and turn it on.  Then it "Removes all Holes" and they wonder what happened.
If you happen to need either one then they come in handy.  Folks just need to remember to turn them off when finished.

MatousRokosVolue, 2022-05-04:
"remove holes" I'd understand but "make overhangs printable".... Well it's hard not to print overhangs if you don't have them in the first place :)

GregValiant, 2022-05-04:
The top of that hole has an overhang / double bevel / chamfer, whatever you want to call it.  I think that's what caused the hole to fill up.  As long as it's sorted out it's all good.