Ultimaker/Cura #12092: v5b1 build plate shrinks with shrinking factor above 100% (horizontal factor)

tispokes, 2022-05-04:
### Application Version

5b1

### Platform

W11

### Printer

SWX1

### Reproduction steps

- Place on build plate
- set horizontal shrinking factor to 150%
build plate will shrink, part extend

### Actual results

smaller build plate

[image]

### Expected results

should stay the same
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

see above

fvrmr, 2022-05-05:
Hi @tispokes thank you for your report.
I will bring this up with the team. Keep you posted.

tispokes, 2022-05-05:
Additional info: I thought/found that I had a big overextrusion when printing in 108% horizontal.

fvrmr, 2022-05-10:
I created a ticket for this, devs see CURA-9271

@tispokes about the overextrusion, I checked this now but couldn't reproduce. So please create a new ticket with a project file.