Ultimaker/Cura #12095: difference in generated gcode between Windows and Ubuntu versions of Cura 4.13.1

BenchmarkCreative, 2022-05-05:
### Application Version

4.13.1

### Platform

Win 7 and ubuntu 20.04

### Printer

Voron V300

### Reproduction steps

Slice the exact same STL, with the exact same profile and printer settings within each Operating system using CURA 4.13.1.  Open the generated gcode and compare.   

### Actual results

There is a visual difference in the top layer of the print, noticeable due to gaps between the top lines
[image]

The g-code is not the same?  Should it be?

### Expected results

I expected that the files would be identical, or off only by a very small differences that are consistent.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Added the 2 gcode files in a zip, wouldn't let me upload them and didn't think anyone would want that copied into here :-)

[PrintQueue.zip](https://github.com/Ultimaker/Cura/files/8628330/PrintQueue.zip)

[image]

BenchmarkCreative, 2022-05-05:
Let me know what additional files you want from each machine?

fvrmr, 2022-05-05:
Hi @BenchmarkCreative thank you for your report.
Did you use the same project file on Windows and Ubuntu?

BenchmarkCreative, 2022-05-05:
I used the same STL... I will use duplicate project file tonight when I return home... would you like it done both ways... i.e. create project on Ubuntu and then use on both machines and then vice versa?
Thanks

BenchmarkCreative, 2022-05-12:
Slicing with same project file yields same gcode... I will need to figure out where the differences were.  Thanks!!!

closed 2022-05-12 by BenchmarkCreative