Ultimaker/Cura #12098: Issue with importing stl files from network drive

tonyjosephm, 2022-05-05:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 3 Extended

### Reproduction steps

1. Can't drag and drop stl files to the application from a network drive.
2. Can't open network drives through the "Open file from" promt.
3. 

### Actual results

1. Can't drag and drop stl files to the application from a network drive.
2. Can't open network drives through the "Open file from" promt.

### Expected results

1. Be able to drag and drop stl files to the application from a network drive.
2. Be able to open network drives through the "Open file from" promt.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

none

fieldOfView, 2022-05-05:
Could you please include the requested logs? Cura.log can be found via Help -> Show configuration folder.

tonyjosephm, 2022-05-05:
[cura.log](https://github.com/Ultimaker/Cura/files/8629922/cura.log)

closed 2022-05-05 by fvrmr