Ultimaker/Cura #12099: CTD after a few hours

Ashtu109, 2022-05-05:
### Application Version

5.0.0-beta+1

### Platform

win 11 on main win 10 on controller

### Printer

anycbic chiron

### Reproduction steps

I have this version installed on 2 different computers. On both it will crash to desktop after a few hours, On my main Desktop its not a total issue but on the one that I use to run and monitor the prints connected directly to the printer it stops printing when the crash happens and it has done it at the exact same spot 2 time in a row. 

I still have the previous normal release installed in a different folder and do not run both at the same time. 

If you are wondering why I use 2 PC's My main is far more powerful than the other so I can setup and slice madel faster and better then off to the cloud and then to my workshop where the printers and other pc are. Doing the SD shuffle is a pain.

### Actual results

n/a

### Expected results

n/a

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8631057/cura.log)

fieldOfView, 2022-05-05:
Could you try if the test release in this comment fixes the crash during printing?
https://github.com/Ultimaker/Cura/issues/11896#issuecomment-1114661108

Ashtu109, 2022-05-05:
maybe I wasn't clear it also crashes on the PC not connect to any printer as well as the one that is after a few hours

MariskaMaas, 2022-05-23:
Hey Ashtu. 
Did you try printing with the fix mentioned by @fieldOfView? Did it work for you?
The issue described crashes both on the PC and on the printer. 
You can also find the fix in 5.0 Stable. 

If the fix did not work, please let us know and we can have a closer look at your case.

closed 2022-06-06 by github-actions[bot]