Ultimaker/Cura #12100: Ugly seams in 5.0 beta standard print settings. Not smart-hiding?

Zanderama, 2022-05-05:
### Application Version

5.0 beta

### Platform

Windows 11

### Printer

Creality Ender 3 V2

### Reproduction steps

Congrats on the 5.0 beta.

However, I found printing the attached simple quarter-sphere results in quite ugly seams in 5.0 beta compared to 4.13.1.  Using the "Standard" print settings on both.  It seems (pun not intended ;) ) that 4.13.1 does its "smart hiding" much better than 5.0 beta, and possibly 5.0 beta is not really trying to hide the seam at all.  

I've not tried printing with "hide seams" rather than the default "smart hiding" yet, in case that helps

[Seam Tester.zip](https://github.com/Ultimaker/Cura/files/8631085/Seam.Tester.zip)

### Actual results

In the images below, Cura 4 is on the left, and Cura 5 is on the right.  You can see Cura 5's lumpy seam on the edge, whereas Cura 4 has tried to hide the seam inside the rear of the object.

[image]
[image]

### Expected results

I'd expect the seams to be more hidden like Cura 4.13.1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

This may relate to the issue #11877 and CURA-8600 I believe mentioned.

sovtek, 2022-05-10:
I also found similar results. If it's helpful, i've attached a comparison of the seams in both versions of cura.

Settings of note:
zseam relative (true)
corner preference (none)
z y seam (110)
z x seam (0)
z seam position (back)
z seam alignment (user specified)
[image]
[image]

Arashrk82, 2022-05-22:
I have the same problem with Cura 5. The Z seam relative is not working like before, the seam line it is not in a straight line anymore.

MariskaMaas, 2022-05-23:
Hey Zanderama, 
Thanks for reporting this issue and the thorough analysis + pictures. 
We are aware of some zseam issue in 5.0. It seems to be related to the new engine in 5.0.
It was also reported in https://github.com/Ultimaker/Cura/issues/11877.
I'm closing this issue as a duplicate, but you can follow the process there.

closed 2022-05-23 by MariskaMaas