Ultimaker/Cura #12102: Cura 5 Beta doesn't support main display on second monitor

theryanmc, 2022-05-05:
### Application Version

5.0.0-beta

### Platform

Windows 10

### Printer

Creality Ender 3 S1 Pro

### Reproduction steps

Set your second monitor as main display
Open cura (you will see the launch window on the not main monitor)
Any pop up windows will also show up on that monitor

### Actual results

Cura won't respect the main monitor preference in windows 10. 

### Expected results

Cura would load popups on the main monitor, or at least on the monitor where Cura is open. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing need.

hawkeyexp, 2022-05-05:
Same effect on windows 11

MariskaMaas, 2022-05-23:
Hi @theryanmc thanks for reporting this.
I will bring this up with the team, and keep you posted.

MariskaMaas, 2022-05-24:
Hey @theryanmc and @hawkeyexp 
It has been discussed with the team and we recognize the behavior you are describing. 
Unfortunately, it has no priority for us to fix or implement this, therefore we decided to defer this. 
I hope you understand.

ScottFerg56, 2022-05-24:
[image]
This bug may be more serious than you know! I've attached a screenshot of the upper-left portion of my primary monitor which shows what I saw when I first ran v5.0. If I close it on my secondary monitor, it reopens with the updated size and position on the 2nd monitor. BUT, **anytime** I close Cura on my primary monitor it reopens this way! Just a sliver that requires some resizing to expose enough of the titlebar to move it to a usable position. (See my other comments on #12140 which was closed and deferred to this bug.)

hawkeyexp, 2022-05-26:
:-( I can understand it's a "cosmetic" issue and functional bugs has to be fixed first but it sucks every day :-)

RonenGru, 2022-05-27:
This is not a cosmetic issue.  On my dual display I have the same behavior.  However, in my system it goes further than just being cosmetic because whenever I open a secondary windows (for example Marketplace) it just opens a blank window on the second monitor and freezes Cura.  I need to shutdown the Cura service to get out of it.  This makes the program completely unusable with two monitors.  In order to make it work I need to disable the secondary monitor.  When I do this the program behaves as it should.  Again, this is not a minor issue! Have not experienced this problem with any other version of Cura or with any other application.

fieldOfView, 2022-05-29:
Is this deferred or closed? Normally a deferred issue is left open.

fieldOfView, 2022-05-29:
This phenomenon is especially problematic on systems where opening a window on the second display results in an unresponsive window (ie: those systems that previously required unchecking the "Restore window position on start" option in the general preferences).

MariskaMaas, 2022-05-30:
Oh! My mistake, I'll open it back up again and bring it up with the team.

Vandresc, 2022-05-31:
Devs see CURA-9344 for our internal reference. 

Thank you all for your feedback!

RonenGru, 2022-06-16:
I think I found the source of this problem.  It's the Nahimic service in Windows...interestingly this was a problem with PrusaSlicer as well.  The Nahimic service is an audio enhancement software that works alongside the system audio driver.  The solution with a dual monitor setup is to disable (or better still stop) the Nahimic service completely.  Unless you have a specific need for it, you probably won't experience any issues disabling it.  The way to disable it is by typing in "services" in the Windows search field (bottom left corner) and opening the services window:
[image]

Right click on the Nahimic service >Properties>Startup type...set to either Manual or Disabled...

Now you'll be able to run Ultimaker Cura 5 on a dual monitor system with freezing or crashing.  RG

ScottFerg56, 2022-06-16:
> I think I found the source of this problem. It's the Nahimic service in Windows...

My system does not have a 'Nahimic' service and is not suffering crashing/freezing problems, so I suspect we have a second issue being conflated with the original one here.

RonenGru, 2022-06-16:
You're right, there are two issues that are apparently unrelated but which seemed to be related originally.  The Cura splash screen still opens on the second monitor and so do any secondary windows (For example, Post Processing Scripts) but with the Nahimic service disabled Cura doesn't crash any more.  With the Nahimic service turned on Cura would inevitably crash.  I'll post the Nahmic solution in the appropriate thread.  Thanks.  RG

nallath, 2022-06-17:
I think that the crashing issue is related to Nahimic. It using the "wrong" screen is a different issue.

Majormer, 2022-06-26:
Odd.  Similar issue, but stopping the Nahimic service seemed to resolve it.  The Nahimic service appears to be an surround sound audio service, so not sure why it would conflict, but it seems to have done the trick.

MagFlux, 2022-07-16:
Just wanted to add that I'm experiencing the same issue closest described by #12562. I have a triple monitor setup and all dialogs plus the splash screen open up on the non-primary monitor while the main application windows opens on the primary monitor. I'd say this should be a higher priority as often I click stuff and nothing happens...until I notice it opened a dialog up elsewhere on my desktop.

rburema, 2022-07-21:
We're still investigating this. (So it we couldn't fix this in 5.1 yet unfortunately.) At present moment, we have two (other) leads, bot largely outside of our sphere of direct influence:

A surprising connection was found to some particular audio drivers (See the comment by @RonenGru above).
Found (and reported to) a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070

EasifyDev, 2022-10-09:
Can confirm that prompts opening on the wrong monitor is a PITA.

I use 3 monitors, main one is monitor #1 in the middle of the three. 

Usually I have all three monitors in portrait, but when working on design apps such as CURA I swivel monitor #1 into landscape and physically move monitor #2 out of the way behind it.

This means that Cura prompts open on monitor #2 behind my main monitor and I can't see them.

It'd be great if you guys could fix this as it is a worse problem than it sounds.

tdwilli1, 2022-12-09:
This is silly that this issue still isn't fixed. It's been over 6 months it looks like. Having the same problem on my system. Have to use Alt-spacebar every time to move the window to the main monitor. Please work on fixing this!

hawkeyexp, 2022-12-26:
Should be addressed and fixed - cura guys please solve it :-)

MariMakes, 2023-03-31:
Quick update on our side for those who are still stuck. 

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download links and the discussions here.](https://github.com/Ultimaker/Cura/discussions/15098)

tdwilli1, 2023-03-31:
Tried reinstalling and got it running.

Opened and closed if on all screens and it opened back up on the last screen that it closed on correctly.

Thanks,
Tim

RonenGru, 2023-03-31:
> I can confirm it is working, too.
>
> Startup splash is always shown on primary display defined by windows and dialog windows from plugins etc. and upload dialog (moonraker plugin) is shown on same display where cura is placed - also if cura is moved after opening from one screen to another!
> Great work - seems the bug is fully solved!

Thank you for this.
I too can confirm that it seems to be working.  The flash screen starts on the right monitor, and from very quick preliminary tests I did all windows open correctly and there are no crashes.
Regards,
Ronen

rburema, 2023-04-04:
Thanks for testing everyone! That's a relief :-)