Ultimaker/Cura #12104: Cura 5 extrusion errors

GT-Geo, 2022-05-05:
### Application Version

5

### Platform

Mac 12.3

### Printer

Ender 3 V2

### Reproduction steps

Sliced a file, portions of the text area have errors. 

The first screen shot is Cura V 4.13.1, the second screen shot is Cura 5. Same settings for both version
[image]
[image]
.

### Actual results

Portions of the text area have errors.

### Expected results

Expect clean results like 4.13.1

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[HH Bust.stl.zip](https://github.com/Ultimaker/Cura/files/8635520/HH.Bust.stl.zip)

GregValiant, 2022-05-06:
When the warning comes up that a model isn't "watertight" that means it has open seams someplace.  That often causes errors in the slice because Cura cannot tell what is the inside and what is the outside.
After having the model repaired I didn't get any problems slicing it.

Here is the report from [formware.](https://formware.co/OnlineStlRepair)
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
--> 17 Non-manifold edges (?)
--> 20 Inverted faces (?)
--> 8 Degenerate faces (?)
--> 0 Duplicate faces (?)
--> 0 Disjoint shells (?)
-> Repairing: 100.00%
----- Repair completed in 10255ms ------
-> Vertex count changed from 32411 to 32390 (-21)
-> Triangle count changed from 64854 to 64806 (-48)

My guess is that when the text was added to the model that the letters weren't imbedded in the model surface and that may have left gaps.

GT-Geo, 2022-05-06:
Thanks for your input, however I tried to repair the file using formware and re-slice it in Cura 5 with the same results
[image]
.

GregValiant, 2022-05-06:
With the model loaded and your settings how you like - use the "File | Save Project" command.  That will create a 3mf file with you printer the model and your settings.  Zip the file and post it here.

GT-Geo, 2022-05-06:
Here's the 3mf file you requested. This is the file after it was run through formware.
[CE3_HH Bust_fixed.3mf.zip](https://github.com/Ultimaker/Cura/files/8639913/CE3_HH.Bust_fixed.3mf.zip)
.

GregValiant, 2022-05-06:
Slicing with your settings in 5.0 I'm not seeing those errors but my video system is in compatibility mode.
Here is a project file with most of the model blocked out.  Try slicing and printing this as a test (it's just a 10 minute print).

[GV_HH Bust.zip](https://github.com/Ultimaker/Cura/files/8641778/GV_HH.Bust.zip)

GT-Geo, 2022-05-06:
Thanks again Greg, slicing it looks to have similar errors. The printed part also is full of those errors. 
[image]
[image]

GregValiant, 2022-05-06:
It's tough to get a good picture.
What I think we're seeing is that the .4 nozzle is too big for the text.  When details fall below what Cura deems to be possible (given the hardware and settings) it "disappears" those features.
I printed that short version on my nicely tuned Ender 3 Pro.
On top is your model.  You can see that "Harry" isn't too bad but that "Houdini" isn't too good.  It's missing pieces and because the letters are so close together there is some stringing making it even fuzzier.
On the bottom is the model scaled at 150%.  Same .4 nozzle but now the features that were gone from the small one are wide enough to print.  (maybe could have picked a better color filament but the silky silver was in the printer).
[image]

Somebody from the Cura team will look at this when they get a chance.  What you could do in Cura is to go to a .2 nozzle and .2 line width and look at the preview of your model and see if those missing areas show up.

GT-Geo, 2022-05-06:
This was sliced with Cura 4.13.1 and printed super clean with a .4 nozzle.  I can't this result with Cura 5.

[image]

GregValiant, 2022-05-06:
Did you try saving it as a project in 4.13 and opening the project in 5.0?  If it's one of the new wall settings in 5.0 that is out of adjustment then the project idea won't fix it because it won't affect it.
I did notice that in the 5.0 project you didn't have Print Thin Walls turned on.  Was it on in 4.13?

GT-Geo, 2022-05-07:
Yes I just tried that, I saved it as a project in 4.13.1 then opened it in 5.0. It shows the same errors. 

Print Thin Walls is the same for both versions, not checked.

MariskaMaas, 2022-05-24:
Hey @GT-Geo, thanks for reporting your issue.

@GregValiant is right, thanks Greg, Cura 5 handles walls in a different ways. Especially details like your letters are defined in a different way than in Cura 4. If you want to learn more about the changed settings. you can find it here https://support.ultimaker.com/hc/en-us/articles/4792077687068-What-new-print-settings-are-introduced-to-Cura-5-

I believe most of the trouble you have been experiencing come from the white zseam/retraction parts in your print preview. They dissapear for me if I reduce the Minimum Wall Line Width to 0.24 mm. This will be smaller than your nozzle, but it's still in the acceptable range so you might achieve the quality you are looking for. 

Can you try to change the Minimum Wall Line Width and let me know if this works for you? 
If not we can have a look at some of the other settings that can be changed to help you. 

Good luck!

[image]