Ultimaker/Cura #12108: Axis z Wont go down pass 5.0

Lela29, 2022-05-06:
### Application Version

unknown 

### Platform

Creality box and creality cr20 pro 

### Printer

creality cr20 pro 

### Reproduction steps

Hi, I Have a Cr 20pro and since I got it  I had some trouble printing on it, I reach out to matterhackers were i purchase the printer from but they told me that since they don't support the creality box they recommend that I reach out to after buy support. 
The printer has not been able to print anything since the axis Z won't go down harder than 5.0mm I have try everything I can think off i have download creality slicer and nothing, at first all 0 out fine like is goin to print well but it don't mater if I intent to print from the TF card or the creality box it won't align that axis to the  -1.08mm I need it on to be able to print again the axis won't respond i have pause the print go to prepare and move axis to that -1.08 and then go back to resume print and it won't accept the change it calibrates and starts over. I have try printing without the TF card on it and only with the TF card on it and not the creality box. At this point I am at a lost. If anyone out there can help I be forever grateful. 

### Actual results

nothing has happen, except my brain and ego are bruise from so much research and little result. 

### Expected results

Help from a super smart person that is way more proficient on technology than I would ever be. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Help Me please

GregValiant, 2022-05-06:
I don't qualify as a Super Smart person but I know that finding out what is really going on can be helpful as well as necessary.
The printer looks to come with an ABL system.  That usually means that there is a setting for Z-offset which is the difference between the ABL sensor "trip point" and where the end of the nozzle actually is in relation to that trip point.  Is the "-1.08" you mentioned your Z-Offset?
That Z-Offset needs to be set in the printer somehow, someway.  Go through the menus on the printer LCD and see if there is a place to input it.
Have you gone to Reddit and searched for a CR20 user group?  The Ender 3 group might have info as well because the Creality printers with factory ABL systems all pretty much all use the same ABL setup.

Stephanos23, 2022-05-08:
I suggest going through methodically to troubleshoot your issue. Sounds like its hardware/firmware related and not Cura.

1. Connect printer to computer and use pronterface or another serial interface
2. Send `G28` to home all axis
3. Send `G0 X0 Y0 Z0` to move it to origin position
4. Check the nozzle position on the printer, is the nozzle on the bed? is it at the corner? Does it pass the paper leveling test? If not then you might want to relevel the bed manually and set the home offsets accordingly. Follow up by sending a `M500` to save settings.

If issues still persist afterwards, you might want to check your start gcode or sliced gcode to see if theres anything that might cause issues there
1. Check if any gcode lines override the offset or tell it to move the Z to absurd heights
2. Check that its using the correct units (mm / inches) 
3. See if bed leveling is correctly enabled, its usually `M420 S1` to enable bed leveling on Creality machines

Check with Creality groups to see if anyone else is having similar issues on your printer model. Creality auto bed leveling firmware usually is a bit weird and sometimes doesn't work correctly like with with CR10 Smart Pro where auto bed leveling doesn't compensate at all.

MariskaMaas, 2022-05-24:
Hey @Lela29, sorry you hear you are stuck.
Unfortunately, this does not seem to be a Cura issue, but a firmware one so I will be closing this issue.

Some Personal Advice; have you tried joining a Creality Owners Discord server? 
They have multiple help channels that might be able to give you the advice you need quickly. 
https://disboard.org/servers/tag/creality

Good Luck!

closed 2022-05-24 by MariskaMaas