Ultimaker/Cura #12110: Checkboxes to change the new scrolling behaviours of menu lists. (Click and Drag Scrolling, Smooth Scrolling)

bridget512, 2022-05-06:
### Is your feature request related to a problem?

Smooth scrolling and Click and Drag Scrolling features have been added to lists in the v5 versions of Cura.

These features are nice for laptops and small devices using touchpads but feels sluggish and slow when scrolling with a mouse wheel.

### Describe the solution you'd like

I'd like to see two checkboxes in the "Preferences > Cura Config > General" section to enable and disable these features independently. 

- Checkbox 1: Enable Smooth Scrolling for lists.
- Checkbox 2: Enable Click and Drag Scrolling in lists.

This would give the user more control over their experience when using the software.

### Describe alternatives you've considered

- Enable scroll smoothing ONLY when click and dragging through the list. When using a mouse wheel the list would behave like previous versions of Cura.

OR

- Heavily reduce the smoothing behavior when scrolling. This would reduce the "sluggishness" felt when using lists.

### Affected users and/or printers

Benefits: 
- Allows users to use the software in a way that does not impact usability or the user experience.
- Faster list scrolling for those using a mouse.

Cons:
- More settings in the preferences menu.

Triangulix, 2022-05-28:
Enforcing interpolated "smooth scrolling" and even more if it simulates inertia is a nuisance to anyone with a mouse wheel.

nallath, 2022-06-10:
I will discuss this with the rest of the team!

nallath, 2022-06-14:
We've added this to the backlog. For our reference: CURA-9382

We do plan to first have a look at the actual behavior of the scrolling (to see if we can get a single behavior that works for all input modes). If we aren't able to figure that out, I think it's fairly likely that we will go for your suggestion. Our main reason for not doing that right away is that we've noticed that a lot of people don't look at the preferences, so we would still get people asking for various behavior changes.

ubershy, 2022-09-08:
Please implement this feature. I want to have option for fast and effective scrolling like in Cura 4. Now it's frustrating.

Rootthecause, 2022-10-15:
I hope this will be still "resolved". In Cura 5.1.1 its still there. This not a bug or a Feature, its pain.

Triangulix, 2022-10-15:
I start to believe they hope that we just get accustomed to the behaviour and stop complaining.

PunaJussi, 2022-10-26:
Cura 5.2.1 and still terrible.

DR-DAT4, 2022-11-21:
honestly, this is the most annoying thing!
Nobody likes it!