Ultimaker/Cura #12111: Cura 5 is a complete desaster

Liep54, 2022-05-06:
### Application Version

5

### Platform

Windows 10 Pro

### Printer

Creality Ender 3 S1

### Reproduction steps

Print a part in Cura 4.13.1 and it was perfect, then printed the same part same settings in Cura 5 and what a desaster.

### Actual results

The printhead will hit the printed part when it retracts to go to a new point to be printed.

### Expected results

Cura 5 still needs a lot of work.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Please fix Cura 5 before releasing it.
[Tardis_Transformer_V2.zip](https://github.com/Ultimaker/Cura/files/8638257/Tardis_Transformer_V2.zip)

rondlh, 2022-05-06:
Cura 5 _beta_ 1 is not a complete "desaster", it offers great improvements, but some bugs need to be ironed out.

GregValiant, 2022-05-06:
What exactly is the problem?
You need to make a proper report or the issues you are having can't be fixed.  I find that 5.0 works very well for my Ender 3 Pro which is nearly identical (slicing wise) as your printer.
Simply providing a laundry list of STL files doesn't help.  Load a file that has a problem, set Cura up to slice, and then use the "File | Save Project" command to create a 3mf file.  Zip the file and post it here.  You could post the gcode file with the "crash" in it.  If the problem is in your "Ending Gcode" that was provided by Creality as part of the definition file then it may not be a Cura problem at all.
You can also post screenshots or take pictures of a print and post them to describe a problem.
When you say "Desaster" must we make an assumption that you meant "Disaster".

Dazzlly, 2022-05-09:
I also had problems printing on my ender 3 (s1), the problem appears when printing via USB the Cura v5 program crashes and closes by itself, I imagine that if I had done it using the SD memory card I would not have this problem but I like use on USB because it monitors temperatures and makes adjustments on the PC and the other version of Cura works normally

nallath, 2022-05-10:
> I also had problems printing on my ender 3 (s1), the problem appears when printing via USB the Cura v5 program crashes and closes by itself, I imagine that if I had done it using the SD memory card [...]

This has already been fixed and is an entirely different issue as reported by the original author (although I'm a bit unclear what the actual issue is that is reported here...)