Ultimaker/Cura #12112: Cura 5 beta 1 incorrect slicing

rondlh, 2022-05-06:
### Application Version

5 beta 1

### Platform

Windows 10

### Printer

Custom

### Reproduction steps

1. Import the model
2. Slice it
3. Check the preview
[image]

### Actual results

Incorrect slicing
[image]

### Expected results

Proper slicing
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I attached the STL, project and created Gcode. 
The model STL parts have 0.6mm width. The used printer nozzle is also 0.6mm. I noticed that the slicing is fine for another printer with a 0.5mm nozzle.

[Circle_test STL 3MF GCODE.zip](https://github.com/Ultimaker/Cura/files/8638723/Circle_test.STL.3MF.GCODE.zip)

rondlh, 2022-05-06:
> That is not an acceptable model for "Spiralize Outer Contour".

Good catch! I didn't mean to use this mode, but still a bit odd.

GregValiant, 2022-05-06:
It needs to be solid all the way through.  A glass full of water to the brim would work.  An empty glass or a half full glass would not.  In the case of the empty glass a slice would pass through the outer wall into solid, then out the inner wall into air, then back into solid and finally back into air.  There would be an inner ring and an outer ring and that would require travel moves to get from one to the other.
I think that's what you are seeing there.  Cura tried to spiralize but there were too many walls and that resulted in a lot of travel moves all over the place and then back to the layer start point.

fvrmr, 2022-05-19:
Hi @rondlh thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/11843, as a workaround you can enable `remove holes`.
I will close this issue, since it is a duplicate.

closed 2022-05-19 by fvrmr