Ultimaker/Cura #12120: SpaceMouse control

handyandyS3, 2022-05-06:
### Is your feature request related to a problem?

I always use a 3DConnexion Space Mouse with my CAD software to design parts for printing.
I then use it with PrusaSlicer to manipulate the model for zooming and viewing.
I get so frustrated that Cura does not support this!
PLEASE can you consider adding this brilliant functionality?

### Describe the solution you'd like

See above

### Describe alternatives you've considered

There are none

### Affected users and/or printers

Everyone with access to a spacemouse.

fieldOfView, 2022-05-06:
There's a plugin for that: https://github.com/smartavionics/RawMouse

handyandyS3, 2022-05-06:
Hi Aldo,
Thanks for that. I have loaded that into the Cura plugins directory and it does work to move the model around.
However, all the axes directions are reversed!
So I'm a bit lost as to what to do to make it work correctly.
Any help would be very much appreciated.
Get BlueMail <https://bluemail.me> for Desktop
Aldo Hoeben wrote:
There's a plugin for that: https://github.com/smartavionics/RawMouse <https://github.com/smartavionics/RawMouse>

handyandyS3, 2022-05-06:
Hi Aldo, Thanks again. I've sorted it!
I actually read the notes right to the end and found the bit about the 'Extensions' menu.
Get BlueMail <https://bluemail.me> for Desktop
Andy Taylor wrote:
Hi Aldo,
Thanks for that. I have loaded that into the Cura plugins directory and it does work to move the model around.
However, all the axes directions are reversed!
So I'm a bit lost as to what to do to make it work correctly.
Any help would be very much appreciated.
Get BlueMail <https://bluemail.me> for Desktop
Aldo Hoeben wrote:
There's a plugin for that: https://github.com/smartavionics/RawMouse <https://github.com/smartavionics/RawMouse>

smartavionics, 2022-05-06:
Yeah, it does help to RTFM!

MariskaMaas, 2022-05-24:
Hi @handyandyS3 

This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/1152 so I'm closing this issue.
I hope the plug-in from @smartavionics and the advice to RFTM helped you on your way. 

Good luck and happy CADing + slicing!

handyandyS3, 2024-11-25:
Please unsubscribe this email account from your mailing list.