Ultimaker/Cura #12123: STILL crashes when chitubox is installed

dishycrafts, 2022-05-06:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

It's a known issue that Cura and Chitubox clash together. With Chitubox installed, Cura tends to crash. For me it happens when I click Save on Disc. 

### Actual results

Cura crashes

### Expected results

It's been over a year FIX IT

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8642603/cura.log)

dishycrafts, 2022-05-06:
[image]

fieldOfView, 2022-05-06:
Have you tried the Cura 5 beta?

fieldOfView, 2022-05-06:
> FIX IT

There's a well-documented workaround. Have you tried that?
https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

dishycrafts, 2022-05-06:
> > FIX IT
> 
> There's a well-documented workaround. Have you tried that? https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

I just found this one. Does option 2 only involve removing the thumbnail? Because option 1 made me facepalm too hard to even consider trying 2 XD

dishycrafts, 2022-05-06:
> Have you tried the Cura 5 beta?

I have not, but I'll give it a try