Ultimaker/Cura #12125: Ability to configure UI font size

truegret, 2022-05-07:
### Is your feature request related to a problem?

The problem is that it is getting tougher to see the small font in the features list/settings.

### Describe the solution you'd like

As an end user I would like to be able to change the font size in the features/settings menu.  Just a few sizes would be desired - 10pt, 12pt, 14pt, 18pt.

### Describe alternatives you've considered

I have changed the resolution of my monitor, but don't believe I want to change it for the features/settings menu in this application and return to the standard high resolution when not using this application.

### Affected users and/or printers

It will affect all users with no impact to any printers.

### Additional information & file uploads

N/A

fieldOfView, 2022-05-07:
Though it is not as easy as a ready-made selection, you can make a custom theme with different font-sizes. Here is an example custom theme to change the font sizes (but keep the rest the same):
https://gist.github.com/fieldOfView/7de3ad07fa06afb821bc0c5fe0b668dc

* download the file using the "Raw" button as "theme.json"
* in Cura, go to Help -> Show configuration folder
* navigate to the "themes" folder, and make a new folder in it named "cura-larger"
* place the "theme.json" inside that folder
* restart Cura
* in the General preferences select "Ultimaker larger" as the theme
* restart Cura once more
* enjoy bigger fonts

Feel free to edit the sizes in the theme.json file.

truegret, 2022-05-07:
Thank you, Aldo.  That was easy!
BTW, I noticed the General Preferences tab has two entries for each "theme".  I have a screenshot if you would like, but basically it (the General|Theme pulldown) looks like this now:
Colorblind Assist Dark
Colorblind Assist Dark
Colorblind Assist Light
Colorblind Assist Light
Ultimaker
Ultimaker
Ultimaker Dark
Ultimaker Dark
Ultimaker Larger

Shall I create a bug with details?  It does not appear to impact functionality ( I selected each, restarted Cura, and checked ).
Cheers

fieldOfView, 2022-05-10:
The doubling of theme entries has already been reported: https://github.com/Ultimaker/Cura/issues/11873

fieldOfView, 2022-05-10:
If this fixes your issue sufficiently, you may close it.