Ultimaker/Cura #12135: Plugin tries to re-install after removal

Doorman352, 2022-05-07:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Longer LK1

### Reproduction steps

1. Uninstalled Octoprint plugin
2. Shutdown computer
3. Next day on opening Cura get a message to sync materials and plugins and Octoprint re-installs.
4. Removed, backed up settings.

### Actual results

The plugin deletes from the list but upon starting Cura it wants to sync and re-install it

### Expected results

Plugin should have been removed and not re-installed 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8645738/cura.log)

Doorman352, 2022-05-25:
> Hi @Doorman352 <https://github.com/Doorman352>
>
> I think I might know what is happening here. The plug-in might be related to your account. What does the blue button say when you visit this page and sign in?
> https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/OctoPrintPlugin
>
>  <https://user-images.githubusercontent.com/40423138/170089538-70f4a8d7-e184-4eda-ad51-37edf347477b.png>
>
> If it says Added to Cura, Cura will install it for you.
> If you click the button again it should change to Add to Cura, the pop-up and sync should stop.
>
> Will you let me know if this works for you?
> Else we will look in a different direction.

Hello,

                After I posted the issue; I loaded the Beta 5.0 and it also tried to sync everything from my account, unsuccessfully. This weekend, I loaded the 5.0 final release and after the first sync my issues resolved and now Octoprint is showing available to install but is not trying to re-install itself after each start of Cura. 

 

                I agree it was probably something in the account, but it would not resolve until after I installed 5.0 final release and then after the initial sync, my account issues have resolved themselves and it did not try to re-install the Octoprint plugin.

 

                Thank you for your time investigating this issue, and I apologize for not posting any update. I was actually really busy using Cura 5.0 and enjoying all the great updates, I haven’t found anything yet that it doesn’t do better. My only problem was that my custom materials did not automatically import, but I was able to get them manually.

 

Thank You

 

 

Sean Parsons (AKA Doorman352)

Ghostkeeper, 2022-06-01:
We'll consider this issue to be resolved then, but it's a good reminder that we need to look at the workflow of syncing plug-ins with the client for cases such as this.

closed 2022-06-01 by Ghostkeeper