Ultimaker/Cura #12136: Sending mesh from Fusion360 to Cura5.0 in inches

MarkJunior88, 2022-05-07:
### Application Version

5.0 beta

### Platform

Windows 11 home

### Printer

Anycubic Kobra Max

### Reproduction steps

1. Clicked bodies
2. Selected the body I wanted and right clicked
3. Saved as mesh
4. Parameters for save as mesh are
     -**Unit type**: inch
     -**Preview mesh**: unchecked
     -**Refinement**: high
     -**Send to 3d Print Utility**: checked
     -**Print Utility**: Custom (C:\Program Files\Ultimaker Cura 5.0.0)
5. Clicked OK

### Actual results

When the part or model comes up in cura it is scaled to 10,000%.
[image]

### Expected results

Normal scale.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

 It works fine with millimeters selected.

MariskaMaas, 2022-05-24:
Hey @MarkJunior88,

Thanks for reporting this. Cura indeed works with mm instead of inches.
If you would like to have the two systems work together you have a plug-in installed that has been contributed by the community. You can find it here https://marketplace.ultimaker.com/app/cura/plugins/nallath/BarbarianPlugin
It will require you to do an extra step in the menu in the top bar when the file is loaded, but you are able to use inches inside of Cura. 

I'm closing this issue since it's already fixed. Please let us know if you are encountering any other issues.

fieldOfView, 2022-05-29:
As an alternative to the manual step required by the Barbarian Plugin, there is also the Mesh Tools plugin, which has an option to automatically interpret models as inches instead of mm.