Ultimaker/Cura #12137: Cura doesnt see printer (anycubic kobra) if another instance of cura is open

jman1118, 2022-05-07:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Anycubic Kobra (custom profile in cura)

### Reproduction steps

Steps to replicate:
1) opened more than one instance of Cura. 
2) Cura is unable to detect printer. 

### Actual results

printer is not showing up in cura.

### Expected results

printer should show up regardless  of instances opened.

 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.txt](https://github.com/Ultimaker/Cura/files/8646007/cura.txt)

GregValiant, 2022-05-08:
If the first instance of Cura takes control of the com port then no other software can access that port.   The only way to get Cura to relinquish control of the port is to close the Cura instance that opened the port.  If you did close the controlling instance of Cura then the automatic port polling of the second instance may find the (now) unused port.  I have not been able to get Cura to control more than a single printer though so even that may not work.

MariskaMaas, 2022-05-24:
Hi @jman1118, 

As @GregValiant shared not being able to operate the same printer in multiple instances of Cura is by design. (Thanks Greg ❤️ ) Why is it important for you to have multiple instances of Cura open? What are you trying to achieve?