Ultimaker/Cura #12138: Print quality issue

JerryCih, 2022-05-08:
### Application Version

5.0.0 Beta

### Platform

Windows 10

### Printer

Creality Ender 4 V2

### Reproduction steps

Done twice, same STL file in Cura 5 and twice the same result.
[spacer_plate_1_-_5mm_v4.zip](https://github.com/Ultimaker/Cura/files/8646119/spacer_plate_1_-_5mm_v4.zip)

### Actual results

STL file sliced in Cura 5.0.0 Beta and printed via Octoprint on Ender 3 V2:
 
[image]
[image]
[image]

### Expected results

Same file sliced in Cura 4.13.1 

[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Cura 5-CE3PRO_spacer_plate_5mm_v4.zip](https://github.com/Ultimaker/Cura/files/8646110/Cura.5-CE3PRO_spacer_plate_5mm_v4.zip)

[Cura 4 CESPRO_spacer_plate_5mm_v4.zip](https://github.com/Ultimaker/Cura/files/8646111/Cura.4.CESPRO_spacer_plate_5mm_v4.zip)

GregValiant, 2022-05-08:
With "Generate Support" turned off there was nothing for the first extrusions of the overhang to squish against and so they just followed the nozzle around.  About the max overhang angle you can get by with without support is 63° and that overhang is 70°.  Whether sliced with 4.13 or 5.0 I think you were fortunate to get it to print correctly at all.

JerryCih, 2022-05-09:
This isnt the first time I'm printing this, never had problems with 4.13 ( 10+ times )

GregValiant, 2022-05-09:
In 5.0, what do you have as the "Wall Ordering" setting?  "Inside to Outside" has some chance of success.  "Outside to Inside" doesn't really as the outside wall would definitely be air-printed in that overhang.
(I've customized my installation and changed the default  to "Inside to Outside" and so I maybe can't tell what your setting is?)

JerryCih, 2022-05-10:
I have "outside to inside". I will try to change and print again

JerryCih, 2022-05-17:
Sorry didnt have time to try to print it again earlier. Changing it to "inside to outside" did the trick. No issue with the orint
[image]
.

closed 2022-05-17 by JerryCih