Ultimaker/Cura #12143: Slicing faults in Cura 4.13.1

AceTanner, 2022-05-08:
### Application Version

4.13.1

### Platform

windows 11

### Printer

ender 3 pro

### Reproduction steps

pulled in the file into Cura and then slice file to send to printer

### Actual results

Cura filled int the details so that there was no hollow box or magnet holes in the sliced file to send to the printer.

### Expected results

hollow box with hinge holes and magnet holes  

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

the following files in zip have before and after slicing
[help faults.zip](https://github.com/Ultimaker/Cura/files/8647344/help.faults.zip)

fieldOfView, 2022-05-08:
Please upload a project file (FIle -> Save project...). That way we can see what settings you use, and tell you which one(s) to change.

fieldOfView, 2022-05-08:
You have probably enabled the "Remove all holes" setting which, surprisingly, removes all holes.

AceTanner, 2022-05-08:
You are right the remove all hole hade been activated thank you for the help. I don’t know how I missed the thanks again.  Sent from Mail for Windows 10 BY Tanner Bailey From: Aldo HoebenSent: Sunday, May 8, 2022 11:24 AMTo: Ultimaker/CuraCc: AceTanner; AuthorSubject: Re: [Ultimaker/Cura] Slicing faults in Cura 4.13.1 (Issue #12143) You have probably enabled the "Remove all holes" setting which, surprisingly, removes all holes.

AceTanner, 2022-05-08:
You are right the remove all hole hade been activated thank you for the help. I don’t know how I missed the thanks again.

fieldOfView, 2022-05-09:
You may close this issue.

closed 2022-05-11 by fvrmr