Ultimaker/Cura #12144: Rotation Tool is not working / Missing circle-Handle

subcamera, 2022-05-09:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1.Import STL-File (smal one)
2. Select Object
3. select rotation tool
4. try to rotate objekt
5. 

### Actual results

no "rotation circles" appeared
rotoation of object ist verx difficult and somtimes imposible

### Expected results

Implementation og "rotation circles" as in version 4

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8649928/cura.log)
[UM2_Kabelkamm.zip](https://github.com/Ultimaker/Cura/files/8649957/UM2_Kabelkamm.zip)

fieldOfView, 2022-05-09:
Thanks for the report. This issue has been flagged before, and has since been fixed for the full release of Cura 5.0.

subcamera, 2022-05-09:
Great. Thanks for reporting. I´m keen on new version :-)

closed 2022-05-09 by nallath