Ultimaker/Cura #12149: Debian 11 - cura can not open files.

LarrySikes, 2022-05-09:
### Application Version

5

### Platform

Debian 11

### Printer

Ender 3 pro

### Reproduction steps

Load Debian 11
Open Cura 5
Try to open a file - actually any file activity does nothing.

### Actual results

Files will not open

### Expected results

File open dialog box.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Fontconfig warning: "/etc/fonts/conf.avail/53-monospace-lcd-filter.conf", line 10: Having multiple values in <test> isn't supported and may not work as expected
Cyclic dependency detected between "file:///tmp/.mount_curauPSROe/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_curauPSROe/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext

tbolender, 2022-05-29:
I can confirm the issue running Ubuntu 20.04. Makes the whole 5 release unusable. Console logging is just slightly different:

```
PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_Ultimat4dS12/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_Ultimat4dS12/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
```

Vandresc, 2022-05-31:
Thank you both for your feedback. We have a ticket on our backlog for further investigation. Duplicate of https://github.com/Ultimaker/Cura/issues/12266

closed 2022-05-31 by Vandresc