Ultimaker/Cura #12154: No matter what i cannot stop the center as my origin. this is a new issue as of today. i tried installing Cura 5 and same thing. i tried adding multiple printers and making sure that in printer setup center is origin is not selected. tried restarting. tried adding the option then removing. nothing works. i have an independent dual extruder printer and usually print 2 parts at a time. i tried slic3r and that is able to work howevere i prefer ultimaker cura

DrCNC, 2022-05-10:
### Application Version

4.11.13 and cura 5.0

### Platform

windows 10

### Printer

custom fff Tronxy IDEX Gemini S (300x300x390)

### Reproduction steps

-restarted application several times after each attempt of a change in the printer settings
-added a new printer with all new settings and never selected center as origin and still when i loaded a part it was still placed automatically in the center with coordinates changed to center as zero even with the axis arrows still at the bottom left corner zero still shows as center. in both older cura and cura 5.0
-tried adding center is origin and then removing it and it sill doesnt go away.( tried this after the issue started but initially center as origin was NOT selected.)
-tried multiple .stl and .3mf files
-used a different slicer and with slic3r parts are able to be placed and identified correctly at bottom left corner as x:0 y:0 z:0

### Actual results

no matter what all parts are sliced at center with wrong coordinates

### Expected results

parts should be sliced at proper location and printed on my printer at the same respective location. please help fix what is happening
[cura_Pawn;prints_wrong_location (1).zip](https://github.com/Ultimaker/Cura/files/8659508/cura_Pawn.prints_wrong_location.1.zip)
.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

i do not know what to send or to show to fix this. when this same issue first happened to me when i first started using ultimaker cura all i had to do was select then un select "use center as origin" in machine settings and  the problem went away. never happened again until now

fieldOfView, 2022-05-10:
For the next time you open an issue on github (for any project): the subject line is not the place to put a long description of what is happening. See it as the subject-line for an email. It should be short and concise. So for example `"use center as origin" option not working` would have been a good subject line.

DrCNC, 2022-05-10:
Who are youSent from Mail for Windows From: Aldo HoebenSent: Tuesday, May 10, 2022 5:10 AMTo: Ultimaker/CuraCc: DrCNC; AuthorSubject: Re: [Ultimaker/Cura] No matter what i cannot stop the center as my origin. this is a new issue as of today. i tried installing Cura 5 and same thing. i tried adding multiple printers and making sure that in printer setup center is origin is not selected. tried restarting. tried adding the option then removing. nothing works. i have an independent dual extruder printer and usually print 2 parts at a time. i tried slic3r and that is able to work howevere i prefer ultimaker cura (Issue #12154) For the next time you open an issue on github (for any project): the subject line is not the place to put a long description of what is happening. See it as the subject-line for an email. It should be short and concise. So for example "use center as origin" option not working would have been a good subject line.

fieldOfView, 2022-05-10:
I am @fieldOfView, one of the top contributors of the Cura project. But who I am is irrelevant; I just gave you a friendly tip on how to improve your (and our) experience on github.

DrCNC, 2022-05-10:
My apologies. I am a member of both communities, GitHub and ultimaker cura but relatively new. I have never posted a topic on github and am not  familiar with how to on either platform.  I thought when I issued that topic this morning that I was sending a support ticket to cura for help with the slicer software\ printer settings issue that I am dealing with. Cura sent a response saying I have to reach out to the company that I bought the printer from which stinks because the issue is only eith cura, when I used  Slic3r’s platform with the same file, it worked and printed fine and at proper coordinates. Anyways, thanks for the heads up! I will head back to youtube for the fix, sorry again for the mixup From: Aldo HoebenSent: Tuesday, May 10, 2022 5:42 AMTo: Ultimaker/CuraCc: DrCNC; AuthorSubject: Re: [Ultimaker/Cura] No matter what i cannot stop the center as my origin. this is a new issue as of today. i tried installing Cura 5 and same thing. i tried adding multiple printers and making sure that in printer setup center is origin is not selected. tried restarting. tried adding the option then removing. nothing works. i have an independent dual extruder printer and usually print 2 parts at a time. i tried slic3r and that is able to work howevere i prefer ultimaker cura (Issue #12154) I am @fieldOfView, one of the top contributors of the Cura project. But who I am is irrelevant; I just gave you a friendly tip on how to improve your (and our) experience on github.

Vandresc, 2022-05-27:
Hi @DrCNC, when using the project your shared, I am not able to reproduce it, the model is not loaded in the center of the buildplate. 
Did you also try a clean install (remove your cura configuration folder, but make sure to do a backup first so not all your profiles will be lost)?

closed 2022-06-10 by github-actions[bot]