Ultimaker/Cura #12156: Printer crash mid print

d1gz0r, 2022-05-10:
### Application Version

5.0-beta

### Platform

Windows 10

### Printer

Ender 3 v2

### Reproduction steps

1. Sliced button for [Wall Mount Key Hanger](https://thangs.com/designer/electrosync/3d-model/Wall-Mount-Key-Hanger-35523) with supports (I have tried tree and gyroid supports).
2. Upload gcode to SD card.
3. Start the print from SD card.

### Actual results

Printer starts laying the first layer after and crashes (stepper motors stop, screen freezes).

### Expected results

Other models from the same SD card prints fine.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Some more details can be found in my Reddit post: [link](https://www.reddit.com/r/3Dprinting/comments/umfc2r/ender_3_v2_crashes_after_few_seconds_of_printing/)
Here is the project file: [Button.zip](https://github.com/Ultimaker/Cura/files/8660899/Button.zip)

fvrmr, 2022-05-11:
Hi @d1gz0r thank you for your report.
Could you also share the gcode file you used? I don't see anything strange from the project file.

d1gz0r, 2022-05-12:
Hello @fvrmr,
Sorry, I do not have it on my PC any more. I will check if it's still on the SD card I am printing from. I will send it here if I find it.

d1gz0r, 2022-05-13:
Hi @fvrmr 
I just got the same bug.
Here is the project and gcode files:
[BeMeGe_cover_R.zip](https://github.com/Ultimaker/Cura/files/8686206/BeMeGe_cover_R.zip)

fieldOfView, 2022-05-13:
This issue has been reported before, and is fixed for the full release of Cura 5.0. See #11896.

d1gz0r, 2022-05-13:
Hello @fieldOfView,
The issue you are referring to is about printing via USB. I am printing from a micro SD card.

GregValiant, 2022-05-14:
I went through that gcode file and there is nothing in there that should affect the printer.  I also printed 20 layers of the file on my Ender 3 Pro and it was fine.