Ultimaker/Cura #12157: Supports not generating in CURA 5 Beta

redmagejr, 2022-05-10:
### Application Version

5.0+1 BETA

### Platform

Windows 10

### Printer

Ultimaker 2+ Extended

### Reproduction steps

1. Added Model
2. Selected Supports
3. Sliced
4. Preview

### Actual results

No Supports were generated even with large overhangs.

### Expected results

Needed supports on model with several islands and overhangs.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fieldOfView, 2022-05-10:
We can't really help you without a project file. You can create a project file via the `File -> Save project...` menu. This will create a file that includes both the model and your settings, and only by having both of those can we tell what is going on and tell you what to change to create the supports.

Asterchades, 2022-05-11:
Looks to me like one of three things:

- You don't have the Support (helper) lines set to show. In other words they're being generated but not pictured.

- Those are under your Support Angle. Just eyeballing it they're going to be around 45°-50°, which is not steep enough for most profile supports to be triggered.

- The Creality profile strikes again. A Minimum Support Area of 2mm² is simply too large for small features like that.

No automatic supports are going to catch that point on the bottom, either. The actual point itself is touching on the bed and the walls are too narrow an angle. A manually added support (either with the Custom Supports plugin or a tweaked Support Blocker, ironically) would be needed to stop that from simply tipping over before it even reached the outcrops.

closed 2022-06-02 by github-actions[bot]