Ultimaker/Cura #12159: Crashes on second time I launch Cura 5

EM-III, 2022-05-10:
### Application Version

5.0.0 Beta

### Platform

Windows 10

### Printer

All Printers

### Reproduction steps

- Install Cura 5 Beta
- Launch Cura 5
- Print successfully
- Exit Cura
- Launch Cura

### Actual results

[image]

### Expected results

Cura Launches correctly ever time

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

If I reinstall Cura 5 beta without uninstalling, it will launch again and I can print successfully. If I close and relaunch, it crashes. I have to run the installer everytime I need to launch. Unsuccessful Launch at 2022-05-10 07:40:26,320 in the log.
[cura.log](https://github.com/Ultimaker/Cura/files/8661424/cura.log)
.

mrzottel, 2022-05-11:
Try starting Cura as administrator. Worked for me, at least as a workaround

fieldOfView, 2022-05-11:
Try it, and report if it works, but it should not be necessary (and is somewhat likely to cause other issues, like for example not being able to drag&drop files into Cura)

closed 2022-05-11 by EM-III