Ultimaker/Cura #12163: Centenas de janelas de erro

josewmc, 2022-05-11:
### Application Version

5.0.0-Beta+1

### Platform

Windows 11

### Printer

Delta Kossel Mini - Marlin 2.0

### Reproduction steps

Após alguns minutos de impressão, aparecem centenas de janelas de erro e eu não consigo ver, travam meu pc.

### Actual results

Não recebi nenhuma sugestão.

### Expected results

Não recebi nenhuma sugestão.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Não sei onde encontrar os logs

fvrmr, 2022-05-11:
Hi @josewmc thank you for your report.
Please fill in the form in English and attach your log file. You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

fieldOfView, 2022-05-12:
This is a duplicate of #11896, which has been fixed for the final release of Cura 5.0.

closed 2022-05-25 by github-actions[bot]