Ultimaker/Cura #12165: Ultimaker Cura-5.0+1 multiple instances crash

MHM-users, 2022-05-11:
### Application Version

5.0+1

### Platform

Windows 10 pro

### Printer

Ultimaker S5 with material station

### Reproduction steps

Tried and failed to make a single copy of an object using the multiply selected. The software crashed after slicing up and down the part in the slicing / print preview.  Multiplying selected items works if the value is larger than 1. It works when the value is 2 or more and then I had to delete one of the instances. This did not crash after restarting but it still did not multiply the selected by 1. 

### Actual results

see above statement, no build blockers were added to the part. 

### Expected results

see above statement

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[UMS5_B958-M0310-7006.zip](https://github.com/Ultimaker/Cura/files/8665885/UMS5_B958-M0310-7006.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8665892/cura.log)

fvrmr, 2022-05-11:
Hi @MHM-users thank you for your report.
The multiplying is a duplicate of https://github.com/Ultimaker/Cura/issues/11894 this is fixed for the 5.0 stable release.
What do you mean with crash? Does your preview crashes? Or does the multiple has an effect on this?

MHM-users, 2022-05-11:
The program closed suddenly and the computer screen went black for a few seconds, the computer recovered and the other software running was not effected. No error windows appeared.  When slicing the part there was a bit of a delay in response just before it quit.

closed 2022-05-25 by github-actions[bot]