Ultimaker/Cura #12166: Marketplace in Cura 5.0 Beta dosent allow scrolling

Cashew92ftw, 2022-05-11:
### Application Version

5.0.0-beta+

### Platform

Windows 11

### Printer

N/A

### Reproduction steps

Selected Marketplace - To see the status of my pluggins
Either from lugins or materials I am not able to scroll down. If I make the window bigger I see more options, but still cant scroll at all.

### Actual results

Cant scroll.

### Expected results

Id expect to see more options which are there since they are cut off when its opened in a smaller window.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I can provide screenshots if needed, but I think this bug is self explanitory.

fieldOfView, 2022-05-11:
I can confirm this on the beta, but not when running from source. That means that either it has been fixed already (and the fix will be part of the final Cura 5.0), or that the scrollbar simply always worked "from source". It hope it is the former.

For now, as a workaround, if you have a scrollwheel or a touchpad, you can scroll without the scrollbar.

fvrmr, 2022-05-11:
Hi @Cashew92ftw thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/12043, so I will close this issue.
It is fixed for the stable released!

closed 2022-05-11 by fvrmr