Ultimaker/Cura #12167: Update notification error.

cmdkill, 2022-05-11:
### Application Version

5.0.0 Beta+1

### Platform

Windows 10

### Printer

CFFFP

### Reproduction steps

[image]

Installed latest Cura 5.0.0 Beta +1 from Github. Ran program at finish of installer. Notified that there was an update available.

### Actual results

Notification popup.
[image]

### Expected results

Notified that update of Beta install completed. No new available.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8666870/cura.log)

No project loaded manually.

4.7 also installed on system.

cmdkill, 2022-05-11:
[cura.log](https://github.com/Ultimaker/Cura/files/8666901/cura.log)

Crash

cmdkill, 2022-05-11:
[cura.log](https://github.com/Ultimaker/Cura/files/8666906/cura.log)
Can't upload gcode file.
Backup and reset cfg.

fieldOfView, 2022-05-11:
Thanks for reporting. This issue has been reported before, and has since been fixed.

fvrmr, 2022-05-11:
Hi @cmdkill thank you for your report.
This is indeed a duplicate of https://github.com/Ultimaker/Cura/issues/11229, so I will close this issue. And this is indeed fix for next beta releases!

closed 2022-05-11 by fvrmr