Ultimaker/Cura #12169: Some Simplified Chinese translation errors

iBral, 2022-05-11:
### Application Version

5.0

### Platform

Win10

### Printer

Voron 2.4

### Reproduction steps

Some Simplified Chinese translation errors, I fixed them. see zip files.
[cura(zh_CN).zip](https://github.com/Ultimaker/Cura/files/8668990/cura.zh_CN.zip)
 

### Actual results

Some Simplified Chinese translation errors, I fixed them. 

### Expected results

Some Simplified Chinese translation errors, I fixed them. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Some Simplified Chinese translation errors, I fixed them.

fvrmr, 2022-05-11:
Hi @iBral thank you for your report.
I will link this to a ticket we have on our backlog!

fvrmr, 2022-05-16:
Thank you for your changes, we updated it and will be included in 5.0!

closed 2022-05-19 by fvrmr