Ultimaker/Cura #12178: Cura 5.0.0 Pre Release Crashes over USB Printing

cfontes, 2022-05-11:
### Application Version

5.0.0-pre-release

### Platform

windows 10

### Printer

Kingroon KP3S 3.0

### Reproduction steps

Load https://www.thingiverse.com/thing:4645349 slice it and try to print over USB.

continue to change between prepera preview views and monitoring view, eventually cura will crash and halt the print.

### Actual results

Curas crashes with multiple reports showing up but all white and you can't see the errors because it happens super fast.

Can I get the logs somewhere to add here?

### Expected results

Cura continues to print over USB like previous versions

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[KK_LegoCalibrationCube.zip](https://github.com/Ultimaker/Cura/files/8672691/KK_LegoCalibrationCube.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8672696/cura.log)

fieldOfView, 2022-05-12:
Thanks for trying the beta, and thanks for reporting this issue. The problem has been reported before, and has since been fixed for the final release of Cura 5. See https://github.com/Ultimaker/Cura/issues/11896

fvrmr, 2022-05-12:
Closing this issue since it is a duplicate.

closed 2022-05-12 by fvrmr