Ultimaker/Cura #12182: The "Print Temperature" limit is too low

Prorok13, 2022-05-12:
### Application Version

Cura Arachne_engine_beta_2 99.10.0

### Platform

win 11

### Printer

Uni3D

### Reproduction steps

I set the "Print temperature" to 390 degrees Celsius

### Actual results

I get the message "unable to slice"

### Expected results

it will be possible to save the G-code

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I use a custom printer. and I print part of it with PEEK plastic with a recommended print temperature of 380-410 degrees Celsius. I would like to be able to set the recommended temperatures from the slicer, and not fix the code manually

fieldOfView, 2022-05-12:
What printer definition do you use in Cura? The printer definition will have to redefine the `maximum_value` for the `material_print_temperature` setting, and a number of other related settings.

Prorok13, 2022-05-12:
I use "Custom FFF printer".  Where can I change these parameters?

fieldOfView, 2022-05-12:
It cannot easily be changed in the GUI, but you have to change .def.json files manually. You could change the `custom.def.json` file for your "Custom FFF Printer", though it would be better if you created an actual printer definition for your printer. You will have to make the following overrides:

```
    "overrides": {
        "default_material_print_temperature": {
            "maximum_value": 500
        },
        "material_print_temperature": {
            "maximum_value": 500
        },
        "material_print_temperature_layer_0": {
            "maximum_value": 500
        },
        "material_initial_print_temperature": {
            "maximum_value": 500
        },
        "material_final_print_temperature": {
            "maximum_value": 500
        },
        "material_break_preparation_temperature": {
            "maximum_value": 500
        },
        "material_break_temperature": {
            "maximum_value": 500
        },
        "material_standby_temperature": {
            "maximum_value": 500
        }
    }
```

Vandresc, 2022-05-27:
Thank you @fieldOfView. Closed in favor of the workaround, reopen if you consider the issue still should be addressed.

closed 2022-05-27 by Vandresc