Ultimaker/Cura #12187: Cura 5.0 Beta - support blocker added on viewpoint instead of object

KimmoHop, 2022-05-13:
### Application Version

5.0.0-beta+1

### Platform

Windows 10 Home 21H2, AMD Ryzen 5 4500U with Radeon Graphics

### Printer

based on Ender 3 Pro

### Reproduction steps

1. Add part on platform
2. Select part
3. Add support blocker on part

### Actual results

Support blocker is added to point of view

### Expected results

Support blocker is added to the part

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

I'll add project file (and possibly log file) when I get to the other PC ;)

fvrmr, 2022-05-19:
Hi @KimmoHop thank you for your report.
I couldn't reproduce this with the 5.0 stable, could you try that as well?
If this is still happening, please add you log file.

KimmoHop, 2022-05-20:
@fvrmr  Works correctly in 5.0 stable :)

[image]

closed 2022-05-20 by KimmoHop