Ultimaker/Cura #12190: Random Seams are not random

BrainSlugs83, 2022-05-13:
### Application Version

5.0.0-beta1

### Platform

Windows 10 - RTX 2070

### Printer

Creality CR-10

### Reproduction steps

1. Set Z Seam Alignment to Random
2. Slice / Re-slice
3. Notice that the Z Seam is not random, and that re-slicing it, doesn't move them (so it's 100% definitely not random).

### Actual results

This happens on any model you slice, but here is a good example of a "Random" Z Seam Alignment in Cura -- it's a straight line for half the model.  And re-slicing doesn't move any of the points, or even the location of the line.

### Expected results

I expect when I chose random, that every layer should have a random z seam position.

For most models, it's statistically unlikely that any two layers would end up aligned (let alone most of them).

Definitely shouldn't see clumps or lines up the side of a model like this.

I also expect, that if it's actually random, then re-slicing it should produce a different result.
(If it needs to be "deterministically pseudo-random" then give us a setting to set the SEED, so that we can change it.)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

It's especially bad with fuzzy skin -- but it's not great without it either.  -- With fuzzy skin the setting is basically ignored; and without it, it picks like between maybe 5 points on the model, instead of it being actually random.  (See images below.)

[image]

[image]

And again, if "determinism" is a requirement of this feature, then please expose the seed so we can change it.  Because, even if you roll a die, if it just lands on 4 every time, that's not random anymore.
[image]

Vandresc, 2022-05-31:
Devs see CURA-9343 for our internal reference. 

Thank you for your feedback @BrainSlugs83!

nallath, 2022-09-08:
Fixed by https://github.com/Ultimaker/CuraEngine/pull/1729

closed 2022-09-08 by nallath

BarsMonster, 2025-01-27:
In Cura 5.9 I still see it is not random.
On square part - it very much clumps up in single corner. 
On cylindrical part - it covers 1/3 of the circumference, and still sometimes stack up exactly at the same position which makes seam very visible.

[image]

[image]

M-Morren221, 2025-04-03:
Same for me. I'm trying to create TPU rollers, and I'm trying to make the seam less noticeable. It would be nice for the Z seam to be completely random.
[image]

Ritbulau, 2025-08-20:
Still present in 5.10.2

[image]