Ultimaker/Cura #12191: Include User Notes in 3mf File

rachael7, 2022-05-13:
### Is your feature request related to a problem?

I tend to save project files (.3mf) for most of my print jobs. I often use non-standard profiles and saving the project file allows me to reprint in the future without having to figure out what I did all over again. However, the project file does not cover everything that might be unique to the project, such as what build plate or glue I used, how the print turned out, etc. A notes section the was saved with the project file would contain this information for me and save me lots of time.

### Describe the solution you'd like

I just want a text box where I can include notes that will be saved with the project file. A text box that was saved with the 3mf file would capture project-specific information for future reference.

### Describe alternatives you've considered

The material definitions have text areas that can store notes about build plate adhesion and print conditions, but those notes apply globally to the material and are not unique to each print. Keeping notes in analog form (notebook, etc) or in a text file would keep the information, but it would not bring it up when I open the project file, so I might not know to look for it.

### Affected users and/or printers

All users and all printers could benefit, though users whose setup changes frequently might find it more helpful than users who are more "point and click".

### Additional information & file uploads

[image]

nallath, 2022-06-15:
I will discuss this feature request with the rest of the team!

NoEzWinz, 2023-08-15:
Any updates on this feature request?