Ultimaker/Cura #12192: Dark Mode Ultimaker Cura.app for Mac only.

mac4kent, 2022-05-13:
### Is your feature request related to a problem?

May 13, 2022
Mac only. Do not use Windows.
[image]

You see General on DARK  is real system you need to put dark mode UltiMaker Cura 
[image]

I really hate too bright light mode this picture 
[image]

Please work on Dark Mode everything. I hate too bright Light Mode and I have low vision and deaf.

### Describe the solution you'd like

I am not good english to explain text. Just look above picture tell you.

### Describe alternatives you've considered

I am not good english to explain text. Just look above picture tell you.

### Affected users and/or printers

I am not good english to explain text. Just look above picture tell you.

### Additional information & file uploads

None

mac4kent, 2022-05-18:
@fvrmr 
Please show me screenshot.

EtzBetz, 2022-11-06:
[image]

mac4kent, 2022-11-06:
There is no Dark Mode. You did not fix Dark Mode feature use System Setting
for Mac only.

mac4kent, 2022-11-06:
You see “Dark” you must put feature Dark Mode I show you picture.

EtzBetz, 2022-11-06:
So you are talking about auto dark mode.
The window border and file picker already respect the system preferences, only the "internal" UI does have its own setting, which doesn't respect system preferences yet.

I created #13743 for auto dark mode as well.

You should learn how to better express your issue and also very much should talk more politely.