Ultimaker/Cura #12195: Beta 5 - Font looks blurry/smeared

jwhite, 2022-05-14:
### Application Version

5.0 B1

### Platform

Windows 11

### Printer

Voron

### Reproduction steps

Loading Cura and viewing on a 4k monitor shows smeared/blurry fonts.

### Actual results

[image]

[image]

### Expected results

Clear fonts.

Cura 4 example :

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

n/a

fieldOfView, 2022-05-14:
It looks like the texts are not aligned on whole pixels. It may be tricky to hunt for the spot where the fractional offset is introduced.

fvrmr, 2022-05-19:
Hi @jwhite thank you for your report. 
This is a known issue, see https://github.com/Ultimaker/Cura/issues/11957. So I will close this issue.

closed 2022-05-19 by fvrmr