Ultimaker/Cura #12197: Dual Monitor Error

tjs2906, 2022-05-14:
### Application Version

5.0 beta 1

### Platform

Windows 11

### Printer

Modified Ender 3 V2 with SKR2

### Reproduction steps

Install software on a computer with dual monitors,
Launch the software
Push the marketplace button

### Actual results

[image]
The marketplace launches on the second monitor causing the software to freeze
The only way to close the program is forced close
I can turn off the second monitor and the software acts as expected

### Expected results

Launch and access the marketplace with both monitors turned on.
Here is a screenshot of single monitor usag
[image]
e

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Feel free to contact me at tjs2906@comcast.net

RonenGru, 2022-05-26:
I'm having the exact same issue.  Pretty much impossible to use Cura 5 until this is fixed.  RG

tjs2906, 2022-05-26:
I just turn off the second monitor and it works fine. But we shouldn't have to do that.

RonenGru, 2022-05-27:
Yes, it does work with one monitor only, but it isn't really a solution that is possible for me to use 99 percent of the time.  I'm wondering if there's any way to force Cura to run on a specific monitor only?  I've been looking for that option in the settings.  If it's there it's not obvious.  Minor point:  you're using Windows 11 and I'm using Windows 10 so it's clearly not just a Windows 11 thing. I'm actually surprised that more people didn't report this problem.

fieldOfView, 2022-05-27:
> I'm actually surprised that more people didn't report this problem.

That is part of the problem: if it were easily reproducable for a dev, it would be a lot easier to create a fix. It is hard to fix something that you cannot "see" happening.

usernametakenaaaa, 2022-05-28:
I don't think there is a second person with dual-monitor who can use Cura 5. This is reproduceable on all of my computers, Cura simply has never been tested with two monitors and they went from beta to release without confirming the hard crash bugreports.

RonenGru, 2022-05-28:
This is not very encouraging since this is not a minor issue.  I too tried  it on two systems with a dual monitor setup and it crashes in both.  In one setup I have an Nvidia GPU and in the other I have a Ryzen.  In both setups Cura 5 does not work with two monitors.

usernametakenaaaa, 2022-05-29:
They re-introduced that bug for half a decade, I found tons of forum posts. 
That also means they, occasionally fix it. 
Basically it is 3 bugs in one:
1) The dialogs are not confined to the proper monitor
2) When the dialog loads on the wrong monitor the application crashes instead of just displaying it
3) The last position of the dialog windows is not stored despite the preference option to store last positions

There are currently two options to run Cura 5.0 in modern environments (more than one screen)
1) Win + P -> disable all screens (this also moves all application windows away from them, you need to re-order them)
2) Install a virtual machine (vmware, virtualbox) and run Cura there. I've not tested that yet. You'll need 3d acceleration support I guess which might complicate it.

tjs2906, 2022-05-29:
I'm sure there are more than just us using dual monitors.  It's kind of a deal breaker for me, I don't want to turn off my second display every time I want to slice something.  And I forget to turn it off before changing settings or going to the marketplace Cura crashes. I realize its free, but?

Vandresc, 2022-05-31:
Hi all, thank you for your feedback. We have a ticket in our backlog for further investigation, devs see CURA-9344 for our internal reference.

nallath, 2022-06-03:
>  Cura simply has never been tested with two monitors and they went from beta to release without confirming the hard crash bugreports.

All of our developers and devs use dual monitors. This is not a matter of us trying to wipe an issue under the carpet, but we honestly didn't see it on any of our machines.

RonenGru, 2022-06-16:
I think I found the source of this problem.  It's the Nahimic service in Windows...interestingly this was a problem with PrusaSlicer as well.  The Nahimic service is an audio enhancement software that works alongside the system audio driver.  The solution with a dual monitor setup is to disable (or better still stop) the Nahimic service completely.  Unless you have a specific need for it, you probably won't experience any issues disabling it.  The way to disable it is by typing in "services" in the Windows search field (bottom left corner) and opening the services window:

[image]

Right click on the Nahimic service >Properties>Startup type...set to either Manual or Disabled...

Now you'll be able to run Ultimaker Cura 5 on a dual monitor system without freezing or crashing.  RG

rburema, 2022-07-21:
We're still investigating this. (So it we couldn't fix this in 5.1 yet unfortunately.) At present moment, we have two (other) leads, bot largely outside of our sphere of direct influence:

- A surprising connection was found to some particular audio drivers (See the comment by @RonenGru  above).
- Found (and reported to) a possible bug in Qt, that might affect certain multi-monitor setups: https://bugreports.qt.io/browse/QTBUG-105070

jldavis1969, 2022-07-31:
> I think I found the source of this problem. It's the Nahimic service in Windows...interestingly this was a problem with PrusaSlicer as well. The Nahimic service is an audio enhancement software that works alongside the system audio driver. The solution with a dual monitor setup is to disable (or better still stop) the Nahimic service completely. Unless you have a specific need for it, you probably won't experience any issues disabling it. The way to disable it is by typing in "services" in the Windows search field (bottom left corner) and opening the services window:
> 
> [image]
> 
> Right click on the Nahimic service >Properties>Startup type...set to either Manual or Disabled...
> 
> Now you'll be able to run Ultimaker Cura 5 on a dual monitor system without freezing or crashing.
> 
> The dialogs are still not confined to the proper monitor, but the software will not crash/freeze. RG

The fix above worked for me. Devs please look into this fix as it may help you finding why Cura 5.0+ is having the issue with multiple monitors in the first place.

Thank you

rburema, 2022-07-31:
@jldavis1969 Thanks for confirming! We are looking into it ... However, other programs or services misbehaving are _very_ hard to work around. It's (currently) not clear what can be done other than complain to them (or us starting to disable other people's services and we _really_ don't want or should to start down that road).

RonenGru, 2022-07-31:
I agree that it's not a  good idea to turn off other people's services, but I think it may be helpful to let people know somehow (perhaps with a pop-up message box when they're installing the program) that the Nahimic service will cause the program to malfunction if they're running a dual monitor setup.  Perhaps it might be possible to trigger the pop up warning only if the Nahimic service is detected on the system.

bryanhunwardsen, 2023-03-27:
I JUST ran into this on a brand new windows build on a brand new tower build. A quick search indicates nahimic is part of asus audio driver installation stack(maybe others), and that it interferes with other applications going back several years (via reddit). Is there any understanding how an Audio service is breaking Cura? And while I dont expect Cura to fix nahimic, is there any possibility of runtime detection, work-around, UI notification, log output (I did not find anything in my cura log but maybe a dev would see it) to alert/warn/direct the user here/to the forum, something? Specifically because the manifestation of this bug issue is so nasty.

FWIW, in troubleshooting, running Cura as admin would allow the application to run, prior to discovery of this thread.

MariMakes, 2023-03-31:
Quick update on our side for those who are still stuck. 

We might have a fix for some of the issues reported here. 🎉
It's a bit hard for us to test, but we have a nightly build available.
Can you please try it out and let us know if it resolves your issue?

[You can find the download link and discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)

edubarr, 2023-05-15:
> I agree that it's not a good idea to turn off other people's services, but I think it may be helpful to let people know somehow (perhaps with a pop-up message box when they're installing the [...]

+1 to a pop up warning about this driver issue. I use both Cura and PrusaSlicer, and the Prusa one when i opened the first time, give me a warning about the audio driver messing something up and maybe the software will behave odd (which didn't happen). Cura just won't load. The fix os disabling Nahimic service is pretty easy at least.

corybsa, 2023-06-04:
> Quick update on our side for those who are still stuck.
>
> We might have a fix for some of the issues reported here. 🎉 It's a bit hard for us to test, but we have a nightly build available. Can you [...]

I just found this thread and using 5.4.0-alpha.1+PyQt64 did not fix the issue for me.

PhotoJoeCA, 2024-02-24:
February 2024 Checking in using 5.4.0 and this is still an issue -- Multi-monitor system - Program hangs on launch.

dshammers, 2024-05-06:
May 2024 and using version 5.4.0 still hangs while launching on 2nd monitor