Ultimaker/Cura #12199: Preview Mode Lags something awful

inspectionsbybob, 2022-05-14:
### Application Version

5.0 Betta

### Platform

Win 10

### Printer

Artist-D

### Reproduction steps

Sliced a large complicated dragon head with tree supports and Lightning infil

### Actual results

Trying to view the preview is near impossible because the screen lags so much.  A simple rotate can take 30sec to a min to respond.  Nevermind trying to scrole through the layers and see what it is doing. 

Watched the CPU performance screen, and the load on CPU0-3 was very uneven.  Not sure if that is Cura or the OS...

### Expected results

Expected much better performance...

[Dragon-Head.zip](https://github.com/Ultimaker/Cura/files/8693657/Dragon-Head.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Enclosed the project file

fvrmr, 2022-05-19:
Hi @inspectionsbybob thank you for your report.
Does this happen for every file? Because for this file I would say that this is expected. Slicing this model took already some time for me.

inspectionsbybob, 2022-05-20:
I thought I had responded, it was stuck in the outbox...

It is the big and complicated models.
But my "other" slicer (Prusa/SuperSlicer sorry) is usable as it lags much less.

I tried a couple of other things, but it is still very very slow, to the point of being unusable.

I WANTED to be able to look at what the supports and infill were doing, but it was too slow, so I had to print a smaller version to see if I should print the big one...

I ended up using SuperSlicer as it would let me scroll through the preview and see where I needed to add an infill modifier.

fieldOfView, 2022-05-20:
Please try if the final 5.0 release, which was published last tuesday, is faster. A performance issue was fixed where there were constant unneccessary screen redraws, which would likely also affect the preview tab.

inspectionsbybob, 2022-05-20:
It is mostly the big & complicated models.

In my "other" slicer (prusa/superslicer sorry) I can sort of use the
preview, it lags but is usable... Your new slicer is MUCH slower.

this model has become my "acid test" for a lot of things as it is
challenging in quite a number of ways... Big, complicated with lots of
vertices, supports, infill, bed adhesion, crossing boundaries, overhangs
and more...

I tried using lightning infill at 60%, and had to wait 2days to see it
fail... it just wasn't enough support under the back of his throat, and
once it caught some of the resulting spaghetti it was over...

The next print I tried was the same model but much smaller so I could see
where/if it was going to fail. Had to eventually put a support modifier in
to pickup the back of the throat properly even using a more conventional
infill.

This is where I really wanted to be able to use the preview to see what the
infill and supports were doing...

What I really wanted was a way to cut the model in half vertically and see
the cross section to see the infill...   XRay mode maybe, haven't played
with that...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

nallath, 2022-05-20:
But can you also validate if the speed has improved in the final release or not?

inspectionsbybob, 2022-05-20:
I will try to download it and try again soon, day-job has me booked...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2022-05-21:
I have been slammed, I will get you an update as soon as I can...

-- 
-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

inspectionsbybob, 2022-06-01:
Sorry for the delay...
I did bring the big dragon in and it is now usable in preview mode.
Sorry for the testing delay....

>
>

closed 2022-06-01 by nallath