Ultimaker/Cura #12201: 0.6mm nozzle slicing error

duncan-rtfm, 2022-05-15:
### Application Version

5.0.0-beta+1

### Platform

Win11

### Printer

Ender 3 Pro

### Reproduction steps

1. Sliced benchy with 0.4 nozzle - all good
2. Sliced with 0.6mm nozzle
    a)  No writing on bottom
    b)  More seriously, the box behind the cabin has a cover!
3. Same anomalies slicing with 0.8 nozzle

### Actual results

0.4mm nozzle:
[image]
0.4mm nozzle:
[image]

0.6mm nozzle:
[image]
0.6mm nozzle:
[image]

### Expected results

I would expect Cura to accurately include all details of the file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I would expect the slicing to show all the features of the project.

GregValiant, 2022-05-15:
Go the the Mesh Tools settings and make sure that Remove all Holes and Make Overhangs Printable are NOT selected.

duncan-rtfm, 2022-05-16:
Hi.  Thank you.  This fixes the 0.6mm, but not the 0.8mm nozzle.  But that's OK.  I'll just go with the 0.6 I think.

Thanks for your help.

Duncan

---------------------------------------
  Duncan Meyer
  Brisbane, QLD
  Email: ***@***.***
  Mob:   0406 527 822

GregValiant, 2022-05-17:
Those settings affect the model.  It shouldn't have mattered what the nozzle size was.  Maybe something in your profile for the .8 nozzle turned those settings back on?

duncan-rtfm, 2022-05-17:
I'll check. Thanks.