Ultimaker/Cura #12203: Build volume incorrectly calculated when part of model is under build plate

jstarrfire, 2022-05-15:
### Application Version

5.0.0

### Platform

MacOS 12.3.1

### Printer

Crealty Ender 3

### Reproduction steps

1. Load a model
2. Scale it larger than the build plate
3. Move model so that only a smaller, printable section is above build plate

### Actual results

Receive "Unable to slice" error. Note: model shadow appears to be based on largest portion below build plate
[image]

### Expected results

The portion of the model under the build plate should be ignored when calculating volume. A printable portion above the build plate should successfully slice.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Reproduced in v4.13.1 and 5.0.0
[eleanora-cut-2-1-bug-report.3mf.zip](https://github.com/Ultimaker/Cura/files/8694506/eleanora-cut-2-1-bug-report.3mf.zip)

jstarrfire, 2022-05-15:
Possibly a duplicate of https://github.com/Ultimaker/Cura/issues/10209

GregValiant, 2022-05-15:
Yes, that won't slice.  There is a workaround if you care to try it.  You need to trick Cura by altering your build plate size and your Home Offsets.  I'll use my Ender as an example.  Home Offsets are stored in M206.
My Home Offset is M206 X0 Y-5 Z0.
My build plate size in Cura is 230 x 230.
By inspecting the Prepare screen in Cura I know that the portion of the model that is above the build plate will fit (skirt and all) and it would print OK, but the portion of the model below the build plate is too big and won't allow Cura to slice.

I Increase my build plate size in Cura to 330 x 330.  That is an extra 50mm all around and now it's big enough that the portion of the model below the build plate fits and the portion above will slice.  The model is located at the midpoint of the Cura build plate.  Slice and save the gcode.
After generating the gcode I open it in a text editor.  Right after the line "Layer:0" I add M206 X50 Y45.  I've added my additional border of 50mm to my X and Y home offsets.
Looking at the end of the gcode file I add M206 X0 Y-5 right before the line ";End of Gcode" to reset my Home Offsets.  Because it's a Creality printer there is a wipe-out move in the end gcode that will want to move the Y to 330 because of the build plate size change ({machine_width} was changed to 330).  I change that 330 back to 230 so it doesn't bang into the endstop.
Then, back in Cura, I reset my build plate size to the correct 230 x 230.

The midpoint of the Cura virtual build plate was changed to X165 Y165.  The midpoint of my actual build plate on the printer is X115 Y115.  By moving my Home Offset 0,0 to the left and forward I've brought the Cura virtual midpoint to my printers actual midpoint and that's where the print will show up.  The Creality purge lines may be an issue if the print will extend that far.  With a skirt setting of 1 line at 1mm from the print you can increase your printable area by a bit.

closed 2022-05-18 by fvrmr