Ultimaker/Cura #12204: I can't export my profiles anymore

Grisoulefifou, 2022-05-15:
### Application Version

5.0

### Platform

Windows10

### Printer

ender 5 plus

### Reproduction steps

I downloaded cura 5.0 and can't upload the profiles

### Actual results

I can find it in the fast access of my windows but it can't find the file and I can't slice models

### Expected results

can't find the show settings in the help menu

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

idmadj, 2022-05-16:
I'm also experiencing this on Windows 11 with 5.0.0 beta+1.

This can probably be consolidated with #12209

This is the output from cura.log:
```
2022-05-16 14:03:01,877 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:01,890 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:01,922 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:01,926 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:01,940 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:01,973 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:02,024 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:02,057 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/PyQt6/Qt6/qml/UM/TooltipArea.qml:0: ReferenceError: Tooltip is not defined
2022-05-16 14:03:07,824 - ERROR - [MainThread] cura.Settings.CuraContainerRegistry.exportQualityProfile [130]: Invalid file format identifier QQuickFileNameFilter(0x2a6b98d8b70)
2022-05-16 14:03:07,824 - WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [431]: file:///C:/Program Files/Ultimaker Cura 5.0.0/share/cura/resources/qml/Preferences/ProfilesPage.qml:368: ReferenceError: folder is not defined```

idmadj, 2022-05-17:
Seems fixed in 5.0.0 public

closed 2022-05-18 by nallath