Ultimaker/Cura #12210: cure 5.0.0 crash after changing printer and selecting manage printers

Onejk1, 2022-05-15:
### Application Version

5.0.0-beta+1

### Platform

windows 10

### Printer

creality ender 3 max & ender 3 V2

### Reproduction steps

1. Change between Printers
2. go to settings and try and select "manage printers"

in the setting dropdown Extruder 1 is shown for how manay times you have swapped printers and as soon as you try and click manage printers it crashes

### Actual results

Crashes

### Expected results

access manage printers

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8695685/cura.log)

Vandresc, 2022-05-27:
Hi @Onejk1, thank you for your feedback. Unfortunately, I am not able to reproduce the issue encountered by you, is this still the case in the stable 5.0? We did a bunch of fixes during beta. 
Many thanks!

Onejk1, 2022-05-27:
Hello

It was only present in the beta, seams to be fine in the stable release 

Steve

closed 2022-05-27 by Vandresc