Ultimaker/Cura #12211: Layer being projected onto baseplate, but gcode seems correct.

mooklepticon, 2022-05-15:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

Load model, slice model, preview layers.  Layer 320 is projected onto the baseplate and is not in the right place. Layer 319 is fine, layer 320 shows up on the baseplate, layer 321 is fine and there's a gap.

Layer 319, everything ok.

[image]

Layer 320 is on the floor. See the thumb hole. Nothing has changed on the top of the model, where 320 should be, but we're actually looking at 319 on the top of the model.

[image]

Layer 321 is fine. We can see a gap at 320

[image]

However, when loaded in an external viewer, it's not present.  https://gcode-viewer.com/

[image]

Here's the STL

[bugged Glasses Wipe Packet holder t200 f102 as text file.txt](https://github.com/Ultimaker/Cura/files/8695750/bugged.Glasses.Wipe.Packet.holder.t200.f102.as.text.file.txt)

### Actual results

Layer 320 is shown on the baseplate despite normal slicing.

### Expected results

Layer 320 should be displayed in the correction position.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Glasses Wipe Packet holder project file.zip](https://github.com/Ultimaker/Cura/files/8695757/Glasses.Wipe.Packet.holder.project.file.zip)

fyi, it seems weird that github takes zips but not rars. 

[cura.log](https://github.com/Ultimaker/Cura/files/8695762/cura.log)

fieldOfView, 2022-05-15:
Looks like a duplicate of the long-standing, impossible to reproduce https://github.com/Ultimaker/Cura/issues/3815

fieldOfView, 2022-05-15:
Turns out https://github.com/Ultimaker/Cura/issues/7412 is the currently "open" variation of that bug.