Ultimaker/Cura #12213: cura keeps closing while printing

kaine518, 2022-05-15:
### Application Version

5

### Platform

windows 11

### Printer

xyz davinci jr 3 in 1 1.0

### Reproduction steps

loaded my project selected my settings and began to print as usual 

### Actual results

the print application closed mid print 

### Expected results

the printer should continue to print 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

none of my files will print all the way through
[CFFFP_Magneto_Helmet Left rear lower.zip](https://github.com/Ultimaker/Cura/files/8696320/CFFFP_Magneto_Helmet.Left.rear.lower.zip)

fieldOfView, 2022-05-16:
Duplicate of #11896

closed 2022-05-16 by nallath

fieldOfView, 2022-05-16:
You read the issue that this is a duplicate of, and find the test-build that includes the fix, all while realizing that you are using beta software.