Ultimaker/Cura #12214: skipping layer

modest-marmot, 2022-05-16:
### Application Version

5.0 b1

### Platform

Windows 10

### Printer

folgertech ft-6

### Reproduction steps

sliced SmartTemperatureTower_PLA_Plus_195-235.stl with .6 nozzle, .2 layer height and slicing produced a skipped layer in the stringing tower - see attached screen shot
smart temperature tower:
https://cdn.thingiverse.com/assets/2a/0b/04/78/c3/SmartTemperatureTower_PLA_Plus_195-235.st

[image]
l

### Actual results

skipped layer at top of cone 4th layer down from top. please see attached project file

### Expected results

slicing to produce geometry for printing at 4th layer down form top of cone

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[FTFT5_SmartTemperatureTower_PLA_Plus_195-235.zip](https://github.com/Ultimaker/Cura/files/8696425/FTFT5_SmartTemperatureTower_PLA_Plus_195-235.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8696429/cura.log)

GregValiant, 2022-05-16:
When you go to a large bore nozzle you lose resolution.  Fine details you can print with a .2 nozzle won't print with a 1.0 nozzle.
That model has errors as well and isn't watertight.
After repairing the model with MS 3D Builder and dropping the line width to .55 it was fine.
[image]

modest-marmot, 2022-05-16:
I tried to replicate your slicing but still get gaps, not sure what i'm
doing wrong, but isn't it intrinsically problematic that it's slicing is
creating a geometry that is floating in space? seems at the very least it
should be flagging this as an issue. I'm including fixed model with ms 3d
builder and bumped wall line widths down to .55 but still getting gapping
in 5.0 and not in 4.12 - see attached screenshot, including new project
file after being processed with ms builder.

Stefan

*Stefan Hampden*
*Principal - CAST architecture*
115-C North 36th Street, Seattle, WA 98103
***@***.***
p. 206.256.9886 x. 101
website <http://www.castarchitecture.com/> : instagram
<https://www.instagram.com/cast.arch/?hl=en> : facebook
<http://www.facebook.com/pages/CAST-architecture/363052273707741?sk=wall>

[image: 2022-05-16_09h41_131.jpg]

closed 2022-06-10 by github-actions[bot]