Ultimaker/Cura #12220: Z Seam at overhangs

ursucristian, 2022-05-16:
### Is your feature request related to a problem?

Is there a way to prevent starting a layer on overhangs? Other than setting your seam manually opposite to the overhanging part.
But what can you do when you have a complicated model with overhangs all over the place?
I have been busy for a while testing on small parts for printing the very popular articulated dragon (https://cults3d.com/en/3d-model/game/articulated-dragon-mcgybeer) and I still have problems with the overhangs of the "horns". I want to make the print as nice as possible thus I must go slow, but at the horns I need to increase the speed so that I get no lifting (at the overhanging part. It looks like I have my settings tuned ok at the moment but I believe the quality would be much better if the seam wouldn't be (sometimes) on the overhang.
[image]
[image]

### Describe the solution you'd like

How about a box we can tick to prevent starting on overhangs?

### Describe alternatives you've considered

Manual definition of the location of the seam. But it doesn't work when the overhangs are all around the model.

### Affected users and/or printers

Everybody

smartavionics, 2022-05-16:
You may care to try my Cura variant that already has this capability. You can find my builds at https://github.com/smartavionics/Cura/releases.

PapsOu, 2022-11-17:
Hello, and what about supporting that feature in the official Cura ?

I think it is a really important feature to avoid starting lines on top of nothing, like that case I have now : 
[image]

luizbgomide, 2023-02-17:
I agree that this is a very desirable feature. In some parts it is impossible to avoid this, even when playing with Seam placement, which ends up in overhangs that curls up or get deformed due to the excess heat.

DejavuDmitry, 2023-10-18:
Yeah, also have this problem

Flopalop2, 2024-01-28:
The new overhang settings prevents this, but would like more flexibility. Right now the seam is placed at the start of the area it deems an overhang so its at a point where flow is changed drastically if you print overhangs slower. Which messes up the seam and overhang.

falkenhawk, 2024-05-14:
> The new overhang settings prevents this

it does not for me. Cura 5.6.0
[image]