Ultimaker/Cura #12221: Addition of ruler in millimeters and centimeters on the table

desenvolvedorze, 2022-05-16:
### Is your feature request related to a problem?

 and a tool that can calculate measurements by selecting part of the part

### Describe the solution you'd like

A tool that can calculate measurements by selecting part of the part, which can be the mouse itself, similar to what happens in Adobe's InDesign. This will facilitate adjustments before printing and aid in accurate resizing.

### Describe alternatives you've considered

improve the resizing tool so that it can display part measurements in detail beyond the full size, e.g. measuring only one side of the part

### Affected users and/or printers

save time and avoid waste

Asterchades, 2022-05-17:
There's a plugin which has this functionality already, including support for Cura 5.0.0 beta 1. It's called "Measure Tool".

fieldOfView, 2022-05-17:
If you have further questions about/suggestions for the Measure Tool plugin, see https://github.com/fieldOfView/Cura-MeasureTool/